nfcim/flutter_nfc_kit #183: FeliCa IDm cannot be obtained.

HpYama, 2024-09-20:
NFCTag tag = await FlutterNfcKit.poll();
After polling the tag, I checked tag.id.

In the case of ISO15693 and ISO14443TypeA, all IDs were obtained.
However, in the case of FeliCa, all IDs could not be obtained.

Specifically, FeliCa's IDm is 16 digits, but only 8 digits are stored in tag.id.

Harry-Chen, 2024-09-20:
On Android, there is no API to read the full IDm, only <https://developer.android.com/reference/android/nfc/tech/NfcF#getManufacturer()>.

You can always use raw `Get FeliCa ID` commands via `transceive` to get the full IDm and PMm.

HpYama, 2024-09-20:
Thank you for your quick reply.

I apologize for the lack of explanation.
I am using iOS.

I understand that it is difficult to obtain all FeliCa IDm using flutter_nfc_kit.

Harry-Chen, 2024-09-20:
> Thank you for your quick reply.
> 
> I apologize for the lack of explanation. I am using iOS.

On iOS, we directly read from [currentIDm](https://developer.apple.com/documentation/corenfc/nfcfelicatag/3043776-currentidm). So it might be limitation of iOS app.

> I understand that it is difficult to obtain all FeliCa IDm using flutter_nfc_kit.

Not actually. As I mentioned, you can always send raw commands.

closed 2024-10-03 by Harry-Chen