nfcim/flutter_nfc_kit #186: Compilation issue after upgrading to Android Studio Ladybug | 2024.2.1

larssn, 2024-10-02:
The exact version of Android Studio is `Build #AI-242.21829.142.2421.12409432, built on September 24, 2024`

I can no longer compile a project that uses flutter_nfc_kit.

The compilation error is:
```
Launching lib/main.dart on sdk gphone64 arm64 in debug mode...
main.dart:1

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_nfc_kit:compileDebugKotlin'.
> Inconsistent JVM-target compatibility detected for tasks 'compileDebugJavaWithJavac' (17) and 'compileDebugKotlin' (21).

  Consider using JVM Toolchain: https://kotl.in/gradle/jvm/toolchain
  Learn more about JVM-target validation: https://kotl.in/gradle/jvm/target-validation 

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 3s
Error: Gradle task assembleDebug failed with exit code 1
Exited (1)
```

The funny part is that I've made no changes to the source project, AND am using VSCode to compile it.
I can only assume that the latest Android Studio upgrade has changed the build environment.

Harry-Chen, 2024-10-03:
Maybe Android Studio is using newer Java / Gradle / AGP / Gradle Kotlin plugin? I have no AS installed so I cannot reproduce it now. You can look into the detailed logs and (hopefully) find more specific information.

larssn, 2024-10-03:
The workaround is similar to last time (#154):
But instead adding (to your project build.gradle):
```groovy
subprojects {
    afterEvaluate {
        if (project.plugins.hasPlugin("com.android.application")
                || project.plugins.hasPlugin("com.android.library")) {
            if (project.name == "flutter_nfc_kit") {
                project.android.compileOptions {
                    sourceCompatibility = JavaVersion.VERSION_21
                    targetCompatibility = JavaVersion.VERSION_21
                }
            }
        }
    }
}
```

And to your `android` section in app build.gradle:
```groovy
kotlin {
    jvmToolchain(21)
}
```

Not sure what is going on, why this is needed, and why it's always this plugin, as its gradle settings seem perfectly reasonable.

Also, building with Android studio is possible without any workaround, for some reason.

NiclasLeichsenring, 2024-10-08:
I'm having the same problem here
`Execution failed for task ':flutter_nfc_kit:compileDebugKotlin'.
> Inconsistent JVM-target compatibility detected for tasks 'compileDebugJavaWithJavac' (11) and 'compileDebugKotlin' (17).`

would be cool to have a real fix for this issue and not only this workaround

Bobbysimon24Osti, 2024-10-08:
I'm having the same problem, but when I set those configurations the error become "Unknown Kotlin JVM target: 21". anyone knows why?

> The workaround is similar to last time (#154): But instead adding (to your project build.gradle):
>
> ```groovy
> subprojects {
>     afterEvaluate {
>         if [...]

ZisisKostakakis, 2024-10-15:
I am not able to either build from AS or from terminal
app/build.gradle
`plugins {
    id "com.android.application"
    id "kotlin-android"
    // The Flutter Gradle Plugin must be applied after the Android and Kotlin Gradle plugins.
    id "dev.flutter.flutter-gradle-plugin"
}

android {
    namespace = "com.example.free_style_libre_2_nfc_tag_reader"
    compileSdk = flutter.compileSdkVersion
    ndkVersion = flutter.ndkVersion

    kotlin {
        jvmToolchain(21)
    }

    compileOptions {
        sourceCompatibility = JavaVersion.VERSION_21
        targetCompatibility = JavaVersion.VERSION_21
    }
    kotlinOptions {
        jvmTarget = JavaVersion.VERSION_21
    }

    defaultConfig {
        // TODO: Specify your own unique Application ID (https://developer.android.com/studio/build/application-id.html).
        applicationId = "com.example.free_style_libre_2_nfc_tag_reader"
        // You can update the following values to match your application needs.
        // For more information, see: https://flutter.dev/to/review-gradle-config.
        minSdk = flutter.minSdkVersion
        targetSdk = flutter.targetSdkVersion
        versionCode = flutter.versionCode
        versionName = flutter.versionName
    }

    buildTypes {
        release {
            // TODO: Add your own signing config for the release build.
            // Signing with the debug keys for now, so `flutter run --release` works.
            signingConfig = signingConfigs.debug
        }
    }
}

flutter {
    source = "../.."
}
`

And 

android/build.graddle

`allprojects {
    repositories {
        google()
        mavenCentral()
    }
}

rootProject.buildDir = "../build"
subprojects {
    project.buildDir = "${rootProject.buildDir}/${project.name}"
}

subprojects {
    afterEvaluate {
        if (project.plugins.hasPlugin("com.android.application")
                || project.plugins.hasPlugin("com.android.library")) {
            if (project.name == "flutter_nfc_kit") {
                project.android.compileOptions {
                    sourceCompatibility = JavaVersion.VERSION_21
                    targetCompatibility = JavaVersion.VERSION_21
                }
            }
        }
    }
}

subprojects {
    project.evaluationDependsOn(':app')
}

tasks.register("clean", Delete) {
    delete rootProject.buildDir
}`

ZisisKostakakis, 2024-10-15:
> I'm having the same problem, but when I set those configurations the error become "Unknown Kotlin JVM target: 21". anyone knows why?

I am on the same scenario here. Can't build no matter what.

hemantbeast, 2024-10-17:
For workaround until this gets fixed, manually update the `gradle.properties` & `build.gradle` files in External Libraries -> Flutter Plugins -> flutter_nfc_kit

For `gradle.properties`, change AGP version to 8.5.2

For `build.gradle`:
```gradle
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_17
        targetCompatibility JavaVersion.VERSION_17
    }

    kotlinOptions {
        jvmTarget = '17'
    }
```

zigapovhe, 2024-10-22:
@larssn I've just tried using your plugin in our pipeline after seeing your PR. Imported your plugin with:
```
  flutter_nfc_kit:
    git:
      url: https://github.com/tillty/flutter_nfc_kit 
```

Before using your plugin, we were getting this error:

```
Execution failed for task ':flutter_nfc_kit:compileReleaseKotlin'.
[        ] > Inconsistent JVM-target compatibility detected for tasks 'compileReleaseJavaWithJavac' (21) and 'compileReleaseKotlin' (17).
```

After changing it to your github fork, we got slightly different error, but still: 
```
Execution failed for task ':flutter_nfc_kit:compileReleaseKotlin'.
[        ] > Inconsistent JVM-target compatibility detected for tasks 'compileReleaseJavaWithJavac' (21) and 'compileReleaseKotlin' (1.8).
```
Any ideas?

larssn, 2024-10-22:
Do you have any workarounds forcefully specifying the kotlin toolchain?

Example:
```groovy
kotlin {
    jvmToolchain(21)
}
```

If so, try removing that, as gradle _should_ be able to detect the correct settings to use.

Pummelo65, 2024-10-26:
I can compile without using a slightly different setting in build.gradle both in example/android/app and android/

android {

    namespace 'im.nfc.flutter_nfc_kit'

    compileSdkVersion 34
    
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
    kotlinOptions {
        jvmTarget = JavaVersion.VERSION_1_8
    }

May be we need to specify the 8.8 instead of the 1.7 version for compatibility now.

ZisisKostakakis, 2024-10-26:
Here's how I made it work for my situation. Keep the default options you get from a new project.
```
compileOptions {
        sourceCompatibility = JavaVersion.VERSION_1_8
        targetCompatibility = JavaVersion.VERSION_1_8
    }

    kotlinOptions {
        jvmTarget = JavaVersion.VERSION_1_8
    }
````

In my situation it was a mismatch versioning from Java and Gradle that was causing the issue.

Here for more info.
https://docs.gradle.org/current/userguide/compatibility.html

I also found this dependency manager to work for me for managing gradle and Java versions.
https://sdkman.io/

Harry-Chen, 2024-12-20:
Solved by #191. Note that even it is possible, I do not plan to lower the compatibility and target java version to 1.8, nor min Android SDK to 21 -- that seems too old for today, and brings about aother compatibility issues (https://github.com/nfcim/flutter_nfc_kit/issues/127#issuecomment-1995538473).

closed 2024-12-20 by Harry-Chen

Harry-Chen, 2024-12-20:
Please try v3.6.0-rc.1

jnorkus, 2025-02-26:
> Please try v3.6.0-rc.1

I can confirm 3.6.0-rc.1 fixes the issue.

Harry-Chen, 2025-02-26:
@jnorkus Thanks! We will release 3.6.0 soon.