nfcim/flutter_nfc_kit #189: 15693 tag : No Mifare tag polled issue

Saintqqq, 2024-11-14:
hi i have a issue
my nfc tag type is 15693
`

    if (tag.type == NFCTagType.iso15693) {
              Uint8List data = Uint8List.fromList([0x03, 0x01, 0x00, 0xfe]);
              try {
                await FlutterNfcKit.writeBlock(2, data,
                    iso15693Flags: Iso15693RequestFlags(),
                    iso15693ExtendedMode: true);
                await Future.delayed(const Duration(milliseconds: 200));
              } catch (e) {
                print("Error writing to ISO 15693 tag: $e");
              }
            }

`

i got this message:

Error writing to ISO 15693 tag: PlatformException(406, No Mifare tag polled, null, null)
how can i do?

thanks

Harry-Chen, 2024-11-14:
https://github.com/nfcim/flutter_nfc_kit/blob/277302081ce70d3ab5aec06f1277854887e2f79c/lib/flutter_nfc_kit.dart#L481-L488

Please read carefully.

closed 2024-11-14 by Harry-Chen

kauemurakami, 2025-03-11:
The same as, I can read my card `type: mifare_classic`.
But when I try to write to its block 0, it changes the `type` to `unknow`, and when I try to rewrite it again I get this error because the type is no longer `mifare_classic` but `unknow`.
But i use the ISO 14443A when read.