nfcim/flutter_nfc_kit #196: NDEF Available : false

cgsrnd, 2024-12-27:
Hi team,

I am trying to implement NFC functionality to share data using the example code provided at [Sample Code](https://github.com/nfcim/flutter_nfc_kit/tree/master/example). However, I am encountering the following issue:

Error Message:
**NDEF Available: false**

`Plugin detail: flutter_nfc_kit: ^3.5.2`

```
flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 3.27.1, on macOS 15.1.1 24B91 darwin-arm64, locale en-IN)
[✓] Android toolchain - develop for Android devices (Android SDK version 34.0.0)
[✓] Xcode - develop for iOS and macOS (Xcode 16.2)
[✓] Chrome - develop for the web
[✓] Android Studio (version 2024.1)
[✓] IntelliJ IDEA Community Edition (version 2024.3.1.1)
[✓] VS Code (version 1.96.2)
[✓] Connected device (4 available)
    ! Error: Browsing on the local area network for Vijaykumar’s iPhone. Ensure the device is unlocked and attached with a cable or associated with the same local area network as this Mac.
      The device must be opted into Developer Mode to connect wirelessly. (code -27)
[✓] Network resources
```

I attached
[image]
 the sample screenshot of output.

dangfan, 2024-12-27:
This is because your tag does not contain a valid NDEF record. It is not an error.

closed 2024-12-27 by Harry-Chen

cgsrnd, 2024-12-30:
Thank you for your response.
I am writing the NDEF tag.
For your reference i am adding more screenshot and sample code. 

Sample Code.

Reading tag,

 ```
 try {
    NFCTag tag = await FlutterNfcKit.poll(
      timeout: Duration(seconds: 90),
      readIso18092: true,
    );
    setState(() {
      _tag = tag;
    });
    await FlutterNfcKit.setIosAlertMessage("Working on it...");
    _mifareResult = null;
    if (tag.standard == "ISO 14443-4 (Type B)") {
      String result1 = await FlutterNfcKit.transceive("00B0950000");
      String result2 = await FlutterNfcKit.transceive("00A4040009A00000000386980701");
      setState(() {
        _result = '1: $result1\n2: $result2\n';
      });
    } else if (tag.type == NFCTagType.iso18092) {
      String result1 = await FlutterNfcKit.transceive("060080080100");
      setState(() {
        _result = '1: $result1\n';
      });
    } else if (tag.ndefAvailable ?? false) {
      var ndefRecords = await FlutterNfcKit.readNDEFRecords();
      var ndefString = '';
      for (int i = 0; i < ndefRecords.length; i++) {
        ndefString += '${i + 1}: ${ndefRecords[i]}\n';
      }
      setState(() {
        _result = ndefString;
      });
    } else if (tag.type == NFCTagType.webusb) {
      var r = await FlutterNfcKit.transceive("00A4040006D27600012401");
      print(r);
    }
  } catch (e) {
    setState(() {
      _result = 'error: $e';
    });
  }

  // Pretend that we are working
  if (!kIsWeb) sleep(Duration(seconds: 1));
  await FlutterNfcKit.finish(iosAlertMessage: "Finished!");
},
``` 

Writing Tag

```
if (_records!.isNotEmpty) {
    try {
      NFCTag tag = await FlutterNfcKit.poll(
        timeout: Duration(seconds: 90),
        readIso18092: true,
      );
      setState(() {
        _tag = tag;
      });

      if (tag.type == NFCTagType.mifare_ultralight ||
          tag.type == NFCTagType.mifare_classic ||
          tag.type == NFCTagType.iso15693) {
        await FlutterNfcKit.writeNDEFRecords(_records!);
        setState(() {
          _writeResult = 'OK';
        });
      } else {
        setState(() {
          _writeResult = 'error: NDEF not supported: ${tag.type}';
        });
      }
    } catch (e, stacktrace) {
      setState(() {
        _writeResult = 'error: $e';
      });
      print(stacktrace);
    } finally {
      await FlutterNfcKit.finish();
    }
  } else {
    setState(() {
      _writeResult = 'error: No record';
    });
  }
},

```
Please Help me. I am trying it for long time but I am getting same error. 
Thank you.

[image]
[image]

dangfan, 2024-12-30:
Based on the provided screenshot, it appears that you may be utilizing Type 4 card technology. If this is the case, kindly refer to the T4T specification for further guidance. Alternatively, you may find a simplified explanation at the following link: https://stackoverflow.com/questions/66951988/nfc-4atag-iso-dep-apdu-and-ndef