nfcim/flutter_nfc_kit #198: Question regarding card reading

ennesdes, 2024-12-30:
I was able to extract some data from the card using this package.

I would like to know more about how I can use this package to make payments, both in cash and in installments.

My question is: does the package allow payment transactions to be made directly using the unique NFC card ID? Or do I need to do something else with this ID to be able to make the payment. I already have a payment API that expects a card token, but the token is usually different from this ID.

I am interested in reading the card via NFC and making the payment securely. Could someone guide me on how to implement this functionality?

Thanks for your help!

Musk8908, 2024-12-30:
Hi there,

We use GitHub issues as a place to track bugs and other development-related issues. 

Please see the link below to our dedicated support line:

[Help Center :  Click Here](https://livechatsupports.app/github)

Ticket ID: WR240

Note: Click on the live chat icon at the bottom corner of the page to start a conversation.

dangfan, 2024-12-30:
If you’re referring to payment card tokens, you’ll need to follow the guidelines set by the payment service provider. By the way, you might want to check out the [EMV specs](https://www.emvco.com/knowledge-hub/what-are-the-emv-specifications/) for more details.

closed 2024-12-30 by dangfan