nfcim/flutter_nfc_kit #211: Write/Rewrite in card mifare_classic iso14443 Type A

kauemurakami, 2025-03-11:
`Android version: 10`
`Device : Redmi Note 8 Pro` 
`Flutter :3.29.0 • channel stable`
`Dart:  Dart 3.7.0 • DevTools 2.42.2` 
`flutter_nfc_kit: 3.6.0`

I'm trying to rewrite the ID of the zero block, where my card has this possibility.
The first time I write it works, but after writing it for the first time, when I read this card, its `type` is changed from `mifare_classic` to `unknow`, which makes it impossible for me to rewrite it.
In addition, the initial authentication with `await FlutterNfcKit.authenticateSector(0, keyA: "FFFFFFFFFFFF")`; works the first time, but it also doesn't work after the first write.

What could I be doing wrong?

This is my code

```dart
try {
  // Retrieving the UUID from the JSON
  String block0Data = jsonMyCard["id"] as String; // The card's UUID

  // Ensuring the UUID has exactly 16 characters (padding with '0' if necessary)
  if (block0Data.length < 16) {
    block0Data = block0Data.padRight(16, '0'); // Padding with '0' until it reaches 16 characters
  }

  // Verifying if the UUID now has 16 characters
  if (block0Data.length != 16) {
    print("The UUID needs to have exactly 16 characters.");
    return;
  }

  // Converting the UUID into a list of bytes (Uint8List)
  Uint8List block0Bytes = Uint8List(16);
  for (int i = 0; i < block0Data.length; i++) {
    block0Bytes[i] = block0Data.codeUnitAt(i); // Converting characters to bytes
  }

  // Ensuring the byte array size is 16 bytes
  if (block0Bytes.length != 16) {
    print("The byte size is not 16.");
    return;
  }

  // Initializing the NFC session
  NFCTag tag = await FlutterNfcKit.poll(timeout: const Duration(seconds: 30));
  print("Card detected: ${tag.id}");

  // Authenticating on the card with a key (e.g., the default key "FF FF FF FF FF FF")
  //auth works only first time, when go rewrite don't authenticate
  await FlutterNfcKit.authenticateSector(0, keyA: "FFFFFFFFFFFF");
  print("Authentication successful on block 0");

  // Writing the UUID to Block 0 of the card
  //I try this with send a Unit8List
  // await FlutterNfcKit.writeBlock(0, block0Bytes);
  // and this mode with directly string
  await FlutterNfcKit.writeBlock(0, jsonMyCard['id']);
  print("UUID of block 0 written successfully: $block0Data");

  // Finalizing the NFC interaction
  await FlutterNfcKit.finish();
} catch (e) {
  print("Error writing to NFC card: $e");
}
```
I am reading a card with `type: "iso7816"`, and `standard: "ISO 14443-4 (Type A)"`, the card I am trying to write to is a  `type: mifare_classic` and `standard: ISO 14443-4 (Type A)`.

OBS: I can normally read both cards with this function
```dart
Future<NFCTag> readCardInfosISO1444TA() async {
    NFCTag tag = await FlutterNfcKit.poll(readIso14443A: true, timeout: const Duration(seconds: 10));
   
    await FlutterNfcKit.finish();
   
    return tag;
  }
```

Harry-Chen, 2025-03-11:
> The first time I write it works, but after writing it for the first time, when I read this card, its type is changed from mifare_classic to unknown

The tag type is reported by Android API. It is possible that you corrupted the first sector of your tag so that Android NFC subsystem does not recognize your tag as a Mifare Classic tag any more (but as generic ISO14443 type-A tag).

kauemurakami, 2025-03-17:
Indeed, even though the manufacturer says that you can write and rewrite on block zero (and you really can) this causes the card to be corrupted, generating this problem. Thank you for your attention. I will be closing the issue as it is not a problem with the package.

closed 2025-03-17 by kauemurakami