nfcim/flutter_nfc_kit #212: Why is the Android minSdkVersion set to 26?

SuperMari000, 2025-04-26:
Hello,

First, thank you very much for creating and maintaining this package! I appreciate all the effort you put into it.

I have an application running on Flutter version 3.29.0, with Android minSdkVersion set to 23 and Gradle version 8.11. I currently use the **NFC Manager** library, and everything works fine.
However, support for NFC Manager has recently been discontinued, and I want to replace it with the **Flutter NFC Kit** library. The obstacle I'm facing is the minSdkVersion requirement.
I have over half a million users on **Android 6.0**, and I cannot drop support for them since **NFC is a core** feature of my app.

**Is it possible to lower the minSdkVersion for Flutter NFC Kit to 23?**

Harry-Chen, 2025-04-27:
Hi! It is not infeasible to lower the `minSdkVersion`. Would you mind if I ask some questions regarding your situation:

1. I could not see anything that could lead me to the conclusion "support for NFC Manager has recently been discontinued", either on its GitHub (although not in active development) or pub.dev page. Could you provide some reference?
2. I am quite surprised that you mentioned "I have over half a million users on Android 6.0". From public statistics, we can see SDK 25 and older versions in total only have <0.1% market share, which adds up to roughly 3-4 million. Could you explain your usage a little bit, e.g. is it public or private usage (like industrial devices)?

SuperMari000, 2025-04-27:
Hi, Thanks for the quick reply.

1. There are many issues popping up in Firebase Crashlytics, and I've noticed that some of them have been present for over a year in the NFC Manager library, On Stack Overflow, I've also seen users suggesting switching to Flutter NFC Kit or applying workarounds, I'm not saying the library is completely unsupported, but support has been very slow, and several issues have remained unresolved for a long time.
2. My app has approximately half a million users on Android 7.0 or earlier. NFC is a core feature, and users rely on it for tasks like charging prepaid electricity, water and gas cards, among other general needs.

Finally, this library supports the web, which is an additional advantage for our users.

SuperMari000, 2025-05-04:
Hi @Harry-Chen,

Any good news ?

ChaserVasya, 2025-10-08:
My app is bought by companies and we have to support the device of each of their employees.  We have a user with the Android 7 (api 24). So, I reopen the issue

ChaserVasya, 2025-10-08:
@Harry-Chen