nfcim/flutter_nfc_kit #215: flutterNfcKit.poll issue

Saintqqq, 2025-05-13:
hi 
My block 0 is E2 40 00 01, declaring it as an NDEF tag. However, my block 1 is (00 00 03 FF), which is an incomplete NDEF data format. 

This causes NFCTag tag = await FlutterNfcKit.poll(); 
to fail, as it cannot detect the tag. What should I do? I need to rewrite clean NDEF data to make it work properly.

Although I declared the tag as NDEF in block 0, the data is not necessarily in NDEF format. I still need NFCTag tag = await FlutterNfcKit.poll(); to be able to read the tag.

thanks

Harry-Chen, 2025-11-22:
We are already using the general `NFCTagReaderSession` on iOS. According to my understanding of CoreNFC APIs, there is not much we could do to control its behavior. If CoreNFC thinks it's NDEF, it will try to read & decode NDEF.

robertpoll, 2025-11-28:
As I understand it 0xE2 in the first byte of block 0 signifies an 8 byte CC so your NDEF TLV header starts at block 2. The 8 byte CC looks good for an 8kB tag (maybe ST25DV64?). We can't see the TLV in block 2+ but assume it's invalid as you say. In my use case if I overwrite the TLV with invalid data on iOS poll() still returns the tag and allows a subsequent transceive() to be successfully sent so you should be able to write a valid NDEF record immediately following the poll(). This is assuming you're on iOS. If you're on Android then poll() will try to read the NDEF and you'll get an error - I've just submitted a PR #225 that allows you to disable the NDEF read on Android poll() using readerModeFlags which may fix your problem.