nfcim/flutter_nfc_kit #220: java.lang.NullPointerException : getHistoricalBytes(...) must not be null

brewv, 2025-09-28:
This problem occurs in Samsung phone Galaxy S25，isoDep.historicalBytes is null
E/AndroidRuntime( 7050): java.lang.NullPointerException: getHistoricalBytes(...) must not be null
E/AndroidRuntime( 7050): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$Companion.parseTag(FlutterNfcKitPlugin.kt:135)
E/AndroidRuntime( 7050): 	at im.nfc.flutter_nfc_kit.FlutterNfcKitPlugin$Companion.handleTag(FlutterNfcKitPlugin.kt:57)
E/AndroidRuntime( 7050): 	at com.tpcl.picture.handle.picture_handle.MainActivity.onNewIntent(MainActivity.kt:28)
E/AndroidRuntime( 7050): 	at android.app.Activity.onNewIntent(Activity.java:2495)
E/AndroidRuntime( 7050): 	at android.app.Activity.performNewIntent(Activity.java:9390)
E/AndroidRuntime( 7050): 	at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1652)
E/AndroidRuntime( 7050): 	at android.app.Instrumentation.internalCallActivityOnNewIntent(Instrumentation.java:1673)
E/AndroidRuntime( 7050): 	at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1661)
E/AndroidRuntime( 7050): 	at android.app.ActivityThread.deliverNewIntents(ActivityThread.java:4726)
E/AndroidRuntime( 7050): 	at android.app.ActivityThread.handleNewIntent(ActivityThread.java:4736)
E/AndroidRuntime( 7050): 	at android.app.servertransaction.NewIntentItem.execute(NewIntentItem.java:58)
E/AndroidRuntime( 7050): 	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:60)
E/AndroidRuntime( 7050): 	at android.app.servertransaction.TransactionExecutor.executeNonLifecycleItem(TransactionExecutor.java:174)
E/AndroidRuntime( 7050): 	at android.app.servertransaction.TransactionExecutor.executeTransactionItems(TransactionExecutor.java:109)
E/AndroidRuntime( 7050): 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:81)
E/AndroidRuntime( 7050): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2880)
E/AndroidRuntime( 7050): 	at android.os.Handler.dispatchMessage(Handler.java:107)
E/AndroidRuntime( 7050): 	at android.os.Looper.loopOnce(Looper.java:257)
E/AndroidRuntime( 7050): 	at android.os.Looper.loop(Looper.java:342)
E/AndroidRuntime( 7050): 	at android.app.ActivityThread.main(ActivityThread.java:9579)
E/AndroidRuntime( 7050): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 7050): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:619)
E/AndroidRuntime( 7050): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:929)

Harry-Chen, 2025-11-22:
I think this is a bug from the vendor (Samsung). The Android API [`getHistoricalBytes`](https://developer.android.com/reference/android/nfc/tech/IsoDep#getHistoricalBytes()) does not return null in its signature.

Harry-Chen, 2025-11-22:
Oh sorry, I see the comment: The historical bytes can be used to help identify a tag. They are present only on IsoDep tags that are based on NfcA RF technology. If this tag is not NfcA then null is returned. Will fix this corner case.